Szybet/InkWatchy #1: Missing library

dariotarantini, 2024-01-01:
I cannot compile the firmware, it miss the images.h header
```
In file included from src/main.cpp:1:
src/defines/defines.h:34:10: fatal error: images.h: No such file or directory

****************************************************************
* Looking for images.h dependency? Check our library registry!
```

Szybet, 2024-01-01:
run the generate.sh file - it will convert images and fonts on the fly to header files which you are missing ;)

Szybet, 2024-01-01:
added a few things to the readme about compiling

dariotarantini, 2024-01-01:
hi, thank you. now i cant build fonts 😢 
./convertFonts.sh: riga 8: cd: ../../.pio/libdeps/watchy/Adafruit GFX Library/fontconvert/: File or directory doesnt exist

Szybet, 2024-01-01:
Oh! let me fix that quickly, please wait ;)

dariotarantini, 2024-01-01:
there is no such dependency. in pio folder

Szybet, 2024-01-01:
I know, my fault im fixing that right now

Szybet, 2024-01-01:
done, should work now with the latest commit

dariotarantini, 2024-01-01:
missing book.txt :), maybe you should add some sample book

dariotarantini, 2024-01-01:
also, unicode package should be installed. should be enough to run pip install unicode, but on my system  (arch) I used `sudo pacman -S python-unidecode`

dariotarantini, 2024-01-01:
yeah missing book break compilation as well

Szybet, 2024-01-01:
> missing book.txt :), maybe you should add some sample book

Nah, generating resources without book will fail, but if you disable book in config it will compile fine. It's in the readme

Szybet, 2024-01-01:
> also, unicode package should be installed. should be enough to run pip install unicode, but on my system (arch) I used `sudo pacman -S python-unidecode`

you are free to edit the script and add your os dependent code to install it, contributions are welcome

dariotarantini, 2024-01-01:
it worked! thank you man, really  a nice project. finally I can use my watchy as a kindle! 😄

closed 2024-01-01 by dariotarantini