Szybet/InkWatchy #4: LoRa Watchy.

samuk, 2024-01-31:
Hi we're kicking around an idea for a respin of Watchy that takes[ RAK Wisblock modules.](https://github.com/Open-Smartwatch/open-smartwatch-gps/issues/12#issuecomment-1918694173) & has on-board GPS

Would your firmware be suitable for running under/alongside [Meshtastic ](https://github.com/meshtastic/firmware)

Initial target would be the [RP2040 board](https://store.rakwireless.com/products/rak11310-wisblock-lpwan-module?m=5&h=wisblock-core) as it's already supported under meshtastic

Szybet, 2024-01-31:
Hi
Cool thing
I don't know about Meshtastic, as there is UI for some reason but as I have shown - This firmware is capable of adding many apps / services to it so maybe?
A few parts of the firmware are esp32 dependent, but some ifdefs and it should run

samuk, 2024-01-31:
Nice, Meshtastic also supports the [NRF52 wisblock module too](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.1.4/zephyr/boards/arm/rak4631_nrf52840/doc/index.html) We'd get bluetooth with that which might be fun.

closed 2024-06-08 by Szybet