Szybet/InkWatchy #5: convertFonts.sh output files

RedBugGamer, 2024-05-18:
You mentioned a few times in your code that you only support Debian. I now have generate.sh working on windows except for convertFonts.sh.
I am still working on it but could you please provide me with the generated files from convertFonts.sh in the meantime?

Szybet, 2024-05-18:
I'm moving everything to littlefs to make everything easier, better organised etc. Do you want the old font.h file or the files that go into littlefs?

If you want upstream your windows changes, you need to make the new littlefs version of scripts working. Keep in mind they are not all ready yet.

RedBugGamer, 2024-05-18:
I'm not quite sure what you mean. I just downloaded it yesterday. I think i need the files that go into littlefs.

RedBugGamer, 2024-05-18:
Thanks. I couldn't find font.h being mentioned in the README. I was looking further into the code and i think i actually need font.h, fontBitmaps.bin, fontGlyphs.bin and fontStruct.bin too. I currently can't manage to compile fontconvert and fontDumper.cpp. So i don't have the output from convertFonts.sh.
Is there anything else I might have missed?

Also IMG_COUNT and FONT_COUNT isn't defined anywhere. Should generate.sh also generate files containing these macros?

Szybet, 2024-05-18:
You are in a horrible mess somehow when you need both the old version and the new one. Go to the latest commit, then recheck

RedBugGamer, 2024-05-19:
Yea you're right. I checked but the lastest commit 0d1eeb7 only has the comment "fix some rare bugs &".

I was able to run generate.sh fully now.  However a headerfile containing IMG_COUNT and FONT_COUNT is still missing. The output only indicates errors with convertBooks.sh and packages.sh. Where should these macros be defined?

RedBugGamer, 2024-05-19:
I manually defined these macros with the number of files and now i got it fully compiled and uploaded!

Szybet, 2024-05-19:
Soon I will finish all changes and this issue will not exist

closed 2024-05-19 by RedBugGamer