Szybet/InkWatchy #18: incorrect documentation in https://github.com/Szybet/InkWatchy/wiki/Compiling-Directly-From-Terminal

picarica, 2025-02-06:
sorry i am too lazy to make pull request but there is error in 
https://github.com/Szybet/InkWatchy/wiki/Compiling-Directly-From-Terminal
2. step
docker doesnt use buildx command needs to be changed from 
`docker buildx build resources/tools/other/compile/ --tag InkWatchy/main`
to 
`docker build resources/tools/other/compile/ --tag InkWatchy/main`

Szybet, 2025-02-06:
I think that both options work?

picarica, 2025-02-06:
```
$ docker buildx build 
docker: 'buildx' is not a docker command.
See 'docker --help'
Docker version 27.4.1, build b9d17eaebb55b7652ce37ae5c7c52fcb34194956

```
[i guess it was just moved to alias](https://docs.docker.com/reference/cli/docker/buildx/build/)
both may work, but i think docker build will be LTS

Szybet, 2025-02-07:
Added, Thanks!

closed 2025-02-07 by Szybet