Szybet/InkWatchy #22: [Feature Request] Turn on LED when an alarm is triggered

glaucon1984, 2025-04-01:
I would like to request a feature, if it's possible: Make the LED indicator blink when an alarm is triggered.

I checked the config.h file and I don't think there is an option for it. I have no idea if this can be implemented, but it would be a very cool feature.

Szybet, 2025-04-02:
It's supported but on Yatchy, because on watchies the led is controlled by the charger, not the CPU so it's not possible on watchies

glaucon1984, 2025-04-03:
Any recommendation on where can I buy a Yatchy?

Szybet, 2025-04-04:
https://github.com/Szybet/Yatchy/tree/main/hardware
The quick answer is nowhere, you could wait on the discord server for a new offer, could take idk 2,3,4,5 months