Szybet/InkWatchy #24: how do i write some UI elements?

picarica, 2025-04-16:
so hello i have been trying out working on some alarms, but i have no idea how to work with this firmware yet, nor ink displays
in alarmQuicksetup.cpp i tired adding 
    String timeStr = String(hourNow) + ":" + (minutesNow < 10 ? "0" : "") + String(minutesNow);
    showMessage("Alarm Set", "Will ring at " + timeStr, 2000); // show for 2 seconds

but it doesnt wotk showMessage function doesnt exists, here, what can i use? not sure i get it how it works yet

Szybet, 2025-04-16:
Well such a function doesn't exist, usually for showing a quick text page I use the `textPage()` function

also, you can't add such a message to the setAlarmQuick() function, because pomodoro uses it and the user shouldn't see the message in that scenario, you need to show the additional info (good idea btw) in the sAQ5() functions (I know you changed it in the pull request, but you get the point)

picarica, 2025-05-13:
added it to the pull reqeust

closed 2025-05-23 by picarica