Szybet/InkWatchy #25: for easier  understanding

picarica, 2025-04-18:
so i was bored today i wanted to set easier flashing of new calendar events, for it to be automated af, i came up with this 

`$ docker run --privileged --mount type=bind,source=/dev/,target=/dev/,consistency=consistent --mount type=bind,source=$(pwd)/.platformio/,target=/root/.platformio/,consistency=consistent --mount type=bind,source=$(pwd)/.esp-idf/espressif/,target=/root/.espressif/,consistency=consistent --mount type=bind,source=$(pwd)/,target=/workspaces/InkWatchy/,consistency=consistent --workdir /workspaces/InkWatchy/ -it InkWatchy/main resources/tools/other/calendar/getEvents.sh
`

i think it would benefit if --workdir would be added to documentation, its unecessary step to **cd /workspaces/InkWatchy**

i propose removing 
`cd /workspaces/InkWatchy`
from [here](https://github.com/Szybet/InkWatchy/wiki/Compiling-Directly-From-Terminal#4-compiling-inkwatchy-for-the-first-time)

and adding

`docker run --privileged --mount type=bind,source=/dev/,target=/dev/,consistency=consistent --mount type=bind,source=$(pwd)/.platformio/,target=/root/.platformio/,consistency=consistent --mount type=bind,source=$(pwd)/.esp-idf/espressif/,target=/root/.espressif/,consistency=consistent --mount type=bind,source=$(pwd)/,target=/workspaces/InkWatchy/,consistency=consistent --workdir /workspaces/InkWatchy/ -it InkWatchy/main /bin/bash`

to [here](https://github.com/Szybet/InkWatchy/wiki/Compiling-Directly-From-Terminal#2-setting-up-the-build-image-and-container)

thank you, simple change, but i think worth editing over, also my solution for easier adding of events can be added to wiki too, because i feel that would be regular thing for watchy users if they have calendar enables, but i think thats streching it  too thin

picarica, 2025-04-18:
i would like to image in future compiling from terminal will be aasier t han from vscode, some nice ncurses ui for everything, entering docker compiling and all that jazz. but first thing first

Szybet, 2025-04-18:
Yea sure, good idea, edit the wiki markdown and paste it here for me to copy over (github wikis don't have pull requests)

picarica, 2025-04-22:
how can i download the markdown of the wiki :D, never done that

Szybet, 2025-04-22:
https://raw.githubusercontent.com/wiki/Szybet/InkWatchy/Compiling-Directly-From-Terminal.md

picarica, 2025-04-22:
```
42c42
< docker run --privileged \
---
> docker run --privileged \ 
47c47,48
< -it InkWatchy/main /bin/bash  
---
> --workdir /workspaces/InkWatchy/ \
> -it InkWatchy/main /bin/bash
50c51
< This should put you in an interactive shell. You should see something along the lines of `root@[identifier]:~#` as a prompt. You will need to be in this shell to complete the rest of the steps.
---
> This should put you in an interactive shell. You should see something along the lines of `root@[identifier]:/workspaces/InkWatchy#` as a prompt. You will need to be in this shell to complete the rest of the steps.
79,84c80
< Enter the project workspace via the following command:
< ```shell
< cd /workspaces/InkWatchy
< ```
< 
< And run the following command:
---
> Run the following command:
131c127
< ```
\ No newline at end of file
---
> ```

```

got diff here and here new file 

[Compiling-Directly-From-Terminal-new.md](https://github.com/user-attachments/files/19849103/Compiling-Directly-From-Terminal-new.md)

Szybet, 2025-04-22:
Edited, added
https://github.com/Szybet/InkWatchy/wiki/Compiling-Directly-From-Terminal/_compare/2ddd1d97fa4111acd55e2a6080cd9c02c91dd669...f1421f8a4a6a0c3fd812cf7fdc50029cc6d1c15f
Thanks for contributing :D

closed 2025-04-22 by Szybet