Szybet/InkWatchy #33: getting current date?

picarica, 2025-05-13:
so i  am simple guys using taychron watchface, i was missing a date so i added it, and it was working but recently i am still setting 13.04 date, instead of 13.05, i wonder if i did something incorrectly, or does tiemRTCLocal have wrong date saved? i thought i was being synced via wifi or something, the calendar understands correct date, why does this not ?
 ```

 #define TIME_CORD_X 18
-#define TIME_CORD_Y 119
+#define TIME_CORD_Y 110
+
+#define DATE_CORD_Y (TIME_CORD_Y + 25 )
+#define DATE_CORD_X 64
+#define DATE_FONT getFont("taychron/Mono13")
 
 #if WATCHFACE_12H
 #define AMPPM_X 150
 #define AMPPM_Y 143
 #define AMPM_FONT getFont("taychron/Mono13")
 
+
 String getAMPM(String time24)
 {
     // Extract the hour from the 24-hour time string
@@ -101,9 +106,19 @@ static void showTimeFull()
     currTime = convertTo12HourFormat(currTime);
 #endif
 
+
+
+//prety sure this draws time
     setTextSize(1);
     setFont(TIME_FONT);
     writeTextReplaceBack(currTime, TIME_CORD_X, TIME_CORD_Y, GxEPD_WHITE, GxEPD_BLACK);
+
+// lets try drawing date below it
+    char dateStr[6];
+    snprintf(dateStr, sizeof(dateStr), "%02d.%02d", timeRTCLocal.Day, timeRTCLocal.Month);
+    setTextSize(1);
+    setFont(DATE_FONT);
+    writeTextReplaceBack(dateStr, DATE_CORD_X, DATE_CORD_Y, GxEPD_WHITE, GxEPD_BLACK);
 }

```
showing just diff in **diff --git a/src/ui/watchface/watchfaces/taychron_tay/taychron.cpp b/src/ui/watchface/watchfaces/taychron_tay/taychron.cpp** should be enought for it to be understandable, i also removed moon phase but that is unrelated

picarica, 2025-05-15:
i checked gsr watchface they are refercing some WAtchTime, what even is that ? can i use that :D

```
void WatchyStarfield::drawDate()
{
    WatchyGSR::display.setFont(&Seven_Segment10pt7b);

    int16_t x1, y1;
    uint16_t textWidth, textHeight;

    int day = WatchTime.Local.Day;
    int dayOfWeek = WatchTime.Local.Wday;
    int month = WatchTime.Local.Month;
    int year = WatchTime.Local.Year + WatchyGSR::SRTC.getLocalYearOffset();

```
i tried it 

```
/ lets try drawing date below it
    char dateStr[6];
    snprintf(dateStr, sizeof(dateStr), "%02d.%02d", timeRTCLocal.Day, WatchTime.Local.Month);
    setTextSize(1);
    setFont(DATE_FONT);
    writeTextReplaceBack(dateStr, DATE_CORD_X, DATE_CORD_Y, GxEPD_WHITE, GxEPD_BLACK);
}

```
to my suprise it compiled, but shows 00, thats hmmm, very intresting

Szybet, 2025-05-15:
DO NOT USE GSR STUFF OR THE Ģ̷̖́Ő̵̝̟D̴̛͕Ŝ̷̗͘ ̶̟͇͊̇W̷̨͌I̷̧̯͋L̴̖̝͆L̷̦̯̒ ̴̭̻̆Ȟ̷̢̱̈́Ā̸̞͓V̶͉͙̇Ḛ̷̼̈́̾ ̶̣̾N̶̘͂̐Ỏ̶̮ ̴̯͆̀Ṃ̵̫͝͝È̸͈̪R̵̗̱̆C̷͗ͅY̵̡͙̓ ̶̢̗̀Ü̵͕͑P̵̜̖͒O̶̼̊͘Ṇ̵͓̂ ̷̼̳̾U̴͚͌̚S̸͙͝

Seriously don't, I will never accept such code, it's cursed, it's bad that you even have access to it but that's another topic

;p

some more read: https://github.com/Szybet/InkWatchy/wiki/Watchfaces#gsr-watchfaces

> no time now to debug it myself might later but look up other watchfaces if the problem is there too and how they do it

just do this

picarica, 2025-05-15:
ye i red all your stuff :D really nice stuff i love it, but i am readlly bad ad c++, and this stuff works so far, but i am lookng at it further, managet to get even string working 

```
// lets try drawing date below it
    char dateStr[10];
    int month = WatchTime.Local.Month;
    String monthName = monthShortStr(month + 1);
    snprintf(dateStr, sizeof(dateStr), "%02d.%s", timeRTCLocal.Day, monthName);
    setTextSize(1);
    setFont(DATE_FONT);
    writeTextReplaceBack(dateStr, DATE_CORD_X, DATE_CORD_Y, GxEPD_WHITE, GxEPD_BLACK);
}

```
and ye i wont push it, i probably dont even have right to modify taychron watchface, i just missed date and tried adding it myself, ill try to look into other watchfaces to get it working :D

Szybet, 2025-05-15:
Generally, I nowhere need to use printf stuff (I dislike the syntax, it's complicated) you also don't need to, but you can

You can modify it (**and push it**) on some conditions:
1. You ask the author for permission
2. You add it as a macro define in config.h that is disabled by default (if the author really dislikes it, he can request to remove it, but that would need to be a huge feature, not just a few lines)
3. You modify enough of the watchface it's considered a fork, then you have your own watchface you can modify as you want, it just needs notes everywhere it's a fork (it can be pushed into the main repo)

if those don't satisfy you, simply keep it in your github fork

picarica, 2025-05-15:
okay i did it without watchy code <3

```
// lets try drawing date below it
//    char dateStr[10];
    String day = String(rM.wFTime.Day);
    String month = getMonthName(rM.wFTime.Month);
    String finalStr = day + " " + month;

//    String monthName = monthShortStr(month + timeRTCLocal.Month + 1);
//    snprintf(dateStr, sizeof(dateStr), "%02d.%s", timeRTCLocal.Day, monthName);
    setTextSize(1);
    setFont(DATE_FONT);
    writeTextReplaceBack(finalStr, DATE_CORD_X, DATE_CORD_Y, GxEPD_WHITE, GxEPD_BLACK);

```
ill bug test it and ask taychron for permission probably,

Szybet, 2025-05-15:
format the code with right click -> format document

also after comments there is a random amount of spaces to the code

(just noting, I know it's a sketch)

picarica, 2025-05-15:
oh, i am editing in vin, and TAB just offsets in wrongly compare to rest to the code, not sure where i can setup custom TAB spacing, so just ye i press SPACE bunch of times

closed 2025-05-16 by picarica