Szybet/InkWatchy #47: there needs to be some script to update config.h and add new variables

picarica, 2025-07-23:
```
src/other/watchdogTask/watchdogTask.cpp:15:20: error: 'BACK_PIN' was not declared in this scope
   15 |     if (buttonRead(BACK_PIN) == BUT_CLICK_STATE || buttonRead(MENU_PIN) == BUT_CLICK_STATE || buttonRead(UP_PIN) == BUT_CLICK_STATE || buttonRead(DOWN_PIN) == BUT_CLICK_STATE)
      |                    ^~~~~~~~
src/other/watchdogTask/watchdogTask.cpp:15:63: error: 'MENU_PIN' was not declared in this scope; did you mean 'MENU_PONG'?
   15 |     if (buttonRead(BACK_PIN) == BUT_CLICK_STATE || buttonRead(MENU_PIN) == BUT_CLICK_STATE || buttonRead(UP_PIN) == BUT_CLICK_STATE || buttonRead(DOWN_PIN) == BUT_CLICK_STATE)
      |                                                               ^~~~~~~~
      |                                                               MENU_PONG
src/other/watchdogTask/watchdogTask.cpp:15:106: error: 'UP_PIN' was not declared in this scope
   15 |     if (buttonRead(BACK_PIN) == BUT_CLICK_STATE || buttonRead(MENU_PIN) == BUT_CLICK_STATE || buttonRead(UP_PIN) == BUT_CLICK_STATE || buttonRead(DOWN_PIN) == BUT_CLICK_STATE)
      |                                                                                                          ^~~~~~
src/other/watchdogTask/watchdogTask.cpp:15:147: error: 'DOWN_PIN' was not declared in this scope
   15 |     if (buttonRead(BACK_PIN) == BUT_CLICK_STATE || buttonRead(MENU_PIN) == BUT_CLICK_STATE || buttonRead(UP_PIN) == BUT_CLICK_STATE || buttonRead(DOWN_PIN) == BUT_CLICK_STATE)
      |                                                                                                                                                   ^~~~~~~~
src/ui/manager/managerFunctions.cpp: In function 'void initDebugMenu()':
src/ui/manager/managerFunctions.cpp:134:27: error: 'DEBUG_ITEM_MOTOR' was not declared in this scope; did you mean 'DEBUG_MENU_MOTOR'?
  134 |         buttons[count] = {DEBUG_ITEM_MOTOR, &emptyImgPack, switchMotorDebug};
      |                           ^~~~~~~~~~~~~~~~
      |                           DEBUG_MENU_MOTOR
src/ui/manager/managerFunctions.cpp:134:76: error: no match for 'operator=' (operand types are 'entryMenu' and '<brace-enclosed initializer list>')
  134 |         buttons[count] = {DEBUG_ITEM_MOTOR, &emptyImgPack, switchMotorDebug};
      |                                                                            ^
In file included from src/defines/defines.h:134:
src/defines/../ui/reUse/menu/menu.h:8:8: note: candidate: 'entryMenu& entryMenu::operator=(const entryMenu&)'
    8 | struct entryMenu {
      |        ^~~~~~~~~
src/defines/../ui/reUse/menu/menu.h:8:8: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const entryMenu&'
src/defines/../ui/reUse/menu/menu.h:8:8: note: candidate: 'entryMenu& entryMenu::operator=(entryMenu&&)'
src/defines/../ui/reUse/menu/menu.h:8:8: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'entryMenu&&'
*** [.pio/build/Watchy_2/src/ui/manager/managerFunctions.cpp.o] Error 1
*** [.pio/build/Watchy_2/src/other/watchdogTask/watchdogTask.cpp.o] Error 1
======================================================================= [FAILED] Took 359.95 seconds =======================================================================

Environment    Status    Duration
-------------  --------  ------------
Watchy_2       FAILED    00:05:59.954
================================================================== 1 failed, 0 succeeded in 00:05:59.954 ==================================================================

```

after every update i have issues where some new variables were added, maybe just some simple bash script that checks for variables not values and adds them? or maybe migrade from .h to yaml file? idk but after every update going line by line and adding new variables

Szybet, 2025-07-23:
Oh I didn't translate `DEBUG_ITEM_MOTOR` my fault sorry fix in a minute

picarica, 2025-07-23:
oh my bad, didnt knew it existed, i have not red docs in a while, is this new thing? i tried it it seems fine but i wonder if its accounting for example the custom quick alarms?

Szybet, 2025-07-24:
Well does it work?

picarica, 2025-07-24:
sorry was testing, it and yes the script worked really great, its a bit slow but other than that its good