Szybet/InkWatchy #51: Cannot upload

escopecz, 2025-07-29:
I followed the Wiki docs and got to the point where everything is green but the Upload step doesn't work:
```
 *  Executing task: platformio run --target upload --environment Watchy_2 

Processing Watchy_2 (board: esp32dev; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21/platform-espressif32.zip; framework: espidf, arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Current pio env: Watchy_2
Env name found in inkwatchy env file in /tmp/: Watchy_2
sdkconfig.defaults contents are identical
Env name found in inkwatchy env file in /tmp/: Watchy_2
Rust is already installed, skipping installation.
Features are: none,snake
Pio env in build rust is: Watchy_2
Rust: Watchy 2 and friends
    Finished `release` profile [optimized] target(s) in 1.53s
2.3M    ../lib/librusty.a
2256    ../lib/librusty.a
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (54.3.21) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-dfuutil-arduino @ 1.11.0 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.0 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-mklittlefs-4.0.0 @ 4.0.0 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-ninja @ 1.10.2 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
*** "custom_component" is used to deselect managed idf components ***
[ComponentManager] Created backup of build file
[ComponentManager] Removed component: espressif/esp_hosted
[ComponentManager] Removed component: espressif/esp_wifi_remote
[ComponentManager] Removed component: espressif/esp-dsp
[ComponentManager] Removed component: espressif/network_provisioning
[ComponentManager] Removed component: espressif/esp-zboss-lib
[ComponentManager] Removed component: espressif/esp-zigbee-lib
[ComponentManager] Removed component: espressif/esp_rainmaker
[ComponentManager] Removed component: espressif/rmaker_common
[ComponentManager] Removed component: espressif/esp_insights
[ComponentManager] Removed component: espressif/esp_diag_data_store
[ComponentManager] Removed component: espressif/esp_diagnostics
[ComponentManager] Removed component: espressif/cbor
[ComponentManager] Removed component: espressif/qrcode
[ComponentManager] Removed component: espressif/esp-sr
[ComponentManager] Removed component: espressif/libsodium
[ComponentManager] Removed component: espressif/esp-modbus
[ComponentManager] Removed component: chmorgan/esp-libhelix-mp3
[ComponentManager] Component not found: espressif/esp32-camera
[ComponentManager] Removed component: espressif/esp_modem
[ComponentManager] Session completed with 20 changes
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ off
Found 60 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Rtc_Pcf8563 @ 1.0.3+sha.92e51fa
|-- Time @ 1.6.1+sha.a18e50d
|-- Adafruit GFX Library @ 1.12.0+sha.87e1550
|-- Adafruit BusIO @ 1.16.1+sha.15fbda5
|-- DS3232RTC @ 2.0.1+sha.ef57d34
|-- NTPClient @ 3.2.1+sha.1aea292
|-- ArduinoJson @ 7.1.0+sha.22dd4da3
|-- Arduino_JSON @ 0.2.0+sha.c282cb6
|-- MoonPhasePlus @ 2.0.1+sha.d6944c5
|-- AsyncTCP @ 3.4.1+sha.d0c5b17
|-- ESPAsyncWebServer @ 3.7.7+sha.d9b612a
|-- GxEPD2 @ 1.5.8+sha.1a15932
|-- StableBMA @ 1.6.0+sha.43e49e8
|-- Grafici-GFX @ 4.0.0+sha.fede954
|-- Olson2POSIX @ 1.6.0+sha.e4ff594
|-- SmallRTC @ 2.4.3+sha.36469c0
|-- open-meteo-arduino @ 0.0.0+20250729165817.sha.58a4004
|-- BLE @ 3.2.1
|-- HTTPClient @ 3.2.1
|-- FS @ 3.2.1
|-- LittleFS @ 3.2.1
|-- WiFi @ 3.2.1
Building in release mode
Retrieving maximum program size .pio/build/Watchy_2/firmware.elf
Checking size .pio/build/Watchy_2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  31.2% (used 102392 bytes from 327680 bytes)
Flash: [====      ]  37.3% (used 1566326 bytes from 4194304 bytes)
*** Original Arduino "idf_component.yml" restored ***
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/en/latest/core/installation/udev-rules.html

Auto-detected: /dev/ttyS3
Uploading .pio/build/Watchy_2/firmware.bin

esptool.py v5.0.0-dev1
A fatal error occurred: Could not open /dev/ttyS3, the port is busy or doesn't exist.
Serial port /dev/ttyS3:
(Could not configure port: (5, 'Input/output error'))

*** [upload] Error 2
============================================ [FAILED] Took 26.09 seconds ============================================

Environment    Status    Duration
-------------  --------  ------------
Watchy_2       FAILED    00:00:26.090
======================================= 1 failed, 0 succeeded in 00:00:26.090 =======================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--environment', 'Watchy_2'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```
I am able, however, upload the demo firmware via https://espressif.github.io/esptool-js/ so I don't think there is any problem with the hardware connection. Any ideas?

escopecz, 2025-07-30:
Macbook Air M1, MacOS 15.5

Szybet, 2025-07-30:
Not supported, there is no good way for usb passthrough in macos, anyway, you can add support for it in 2 ways:
1. Write a Mac OS native script which takes the compiled binaries and flashes them, outside of the container
2. Maybe this will work but it's a bit hacky: https://blog.golioth.io/usb-docker-windows-macos/

Meffi29a, 2025-08-06:
Unfortunately, I'm also one of those Mac users. The current beta release 26 even supports containers. Maybe things will improve soon. I'm not starting to install betas (OS & Xcode) right now, though.

I'm quite inexperienced when it comes to shared flashing binaries.
Firmware & File System...
Shouldn't creating a demo image be enough?
I'm trying to flash the created demo file firmware.elf. It's much larger than the provided "Watchy_2-demo.bin." That doesn't seem to be a solution.

I need help!
Which files do I flash and in what order to get a self-compiled version to work?

Better: How do I get a self-compiled demo.bin?
I don't think manual flashing is that bad.

If I can manage that, maybe I can help tweak the firmware in a few small ways. I'm currently considering rebuilding the ESP for Watchy. But it's kind of a dead horse :D

Szybet, 2025-08-06:
Oh yea I forgor about this issue

Now that demo generating is improved, you can just generate one and use it instead

firmware.elf is completly something else, stop, won't work

Ok so smoll tutorial

[image]

[image]

[image]

[image]

github messed up the images, whatever

then files will appear at `resources/tools/other/out/demo`, just flash them as regular demos

choose your device that you have

Meffi29a, 2025-08-22:
I'm currently writing an AppleScript to sync  InkWatchy with the Mac.

I chose the AppleScript because it uses Mac's own tools and is therefore suitable for most sunshine Mac users.
Current progress:
-Download offset.txt, fs.bin, and Watchy_2-demo.bin
-Open offset.txt to read the value.

What I want to achieve:
A dialog window for
-selecting a full sync with a demo.bin (which was previously customized)
-flash the fs with the current offset.
Catch some errors I found on the my Way through. 

I think this could be a good tool for all Mac user. 
Thous have to Instal Brew, brew her esptool. Find the Name of the Container and provide some folder in her Documents-Folder.

Are the more ideas for the Skript?

Szybet, 2025-08-22:
demo already contains a filesystem soo

you could choose what to uploud, filesystem or firmware and just uploud whats needed (So a bit more native, not like the demo)

Look up the demo creating script on how to flash firmware (mostly addresses of bootloader, partition table, firmware binary) etc

Szybet, 2025-08-22:
demo is just an empty file with other "files" (filesystem, bootloader, firmware etc...) written into it at specified offsets. You simply need to make a script that takes those existing files (after compilation/generation) and flashes them at specified offsets in the flash of the watchy.

The demo script is long because it triggers compilation / configures the firmware a bit automatically, you don't need that

Meffi29a, 2025-08-22:
I'm confused. Isn't it what I try to do. Start with a custom Demo and flash it. 
For a "small" update in the fs, flash it at the given offset with "esptool write_flash $offset out/fs.bin"

For a change in the Modules make a new Demo and flash it.

writeAtOffset demo.bin $bootloaderPath $bootloaderAddress 
-will never change

writeAtOffset demo.bin $partitionTablePath $partitionTableAddress
-maybe div but only with changing modules / config.h and confidential.h. 
        Therefor we can build a new Demo and flash

writeAtOffset demo.bin $firmwarePath $firmwareAddress
-the same as partitionTable

writeAtOffset demo.bin $resourcesPath $resourceAddress
-starts always on the address in offset.txt.

Szybet, 2025-08-22:
No, no, no, demo is not a magic file that works for every offset, no, no

This is what a demo is

https://github.com/Szybet/InkWatchy/blob/545a5bed371be94002e12e7aafafbf9046727298/resources/tools/other/compile/demoMaker.sh#L154-L177

It takes these files:
```
    bootloaderPath=".pio/build/$envName/bootloader.bin"
    partitionTablePath=".pio/build/$envName/partitions.bin"
    firmwarePath=".pio/build/$envName/firmware.bin"
    resourcesPath="resources/tools/fs/out/fs.bin"
```

And writes them at the correct offsets of the demo.bin

```
    writeAtOffset demo.bin $bootloaderPath $bootloaderAddress
    writeAtOffset demo.bin $partitionTablePath $partitionTableAddress
    writeAtOffset demo.bin $firmwarePath $firmwareAddress
    writeAtOffset demo.bin $resourcesPath $resourceAddress
```

What you need to do is to create a script that flashes those things outside of the container directly to the device using esptool. Also so to uploud books you won't need to reflash firmware, make the firmware and filesystem seperate to being flashed (and an all option too)

Here is a chatgpt example of what I mean (CLI bash script for linux, idk what macos does)

Also, support for different bootloader offsets is also needed (just like demomaker does it)
```bash
#!/bin/bash

PORT=${PORT:-/dev/ttyUSB0}
BAUD=${BAUD:-921600}

envName=$1
action=$2

bootloaderPath=".pio/build/$envName/bootloader.bin"
partitionTablePath=".pio/build/$envName/partitions.bin"
firmwarePath=".pio/build/$envName/firmware.bin"
resourcesPath="resources/tools/fs/out/fs.bin"
resourceAddress=$(<resources/tools/fs/in/offset.txt tr -d '\n')

bootloaderAddress=0x1000
partitionTableAddress=0x8000
firmwareAddress=0x10000

flash_all() {
    esptool.py --chip esp32 --port "$PORT" --baud "$BAUD" write_flash \
        $bootloaderAddress "$bootloaderPath" \
        $partitionTableAddress "$partitionTablePath" \
        $firmwareAddress "$firmwarePath" \
        $resourceAddress "$resourcesPath"
}

flash_firmware() {
    esptool.py --chip esp32 --port "$PORT" --baud "$BAUD" write_flash \
        $firmwareAddress "$firmwarePath"
}

flash_fs() {
    esptool.py --chip esp32 --port "$PORT" --baud "$BAUD" write_flash \
        $resourceAddress "$resourcesPath"
}

case "$action" in
    all)
        flash_all
        ;;
    firmware)
        flash_firmware
        ;;
    fs)
        flash_fs
        ;;
    *)
        echo "Usage: $0 <envName> {all|firmware|fs}"
        exit 1
        ;;
esac
```

Meffi29a, 2025-08-27:
@escopecz 
could you try to use this script: https://github.com/Meffi29a/InkWatchy/blob/master/resources/tools/other/compile/flashWithMac.AppleScript

 schuld work out of the box while DevContainer is open.