Szybet/InkWatchy #53: When turning on wifi the Inkfield watchface loses the font for the time

TheSimpleZ, 2025-07-30:
As the title states, when turning Wi-Fi on, the time on the Inkfield watch face loses its font. When a redraw happens, the next minute is displayed correctly. Only way to solve it is to go into the menu and back.

Here it is right after Wi-Fi has been turned on:
[image]

Here it is a few minutes after, it's actually 16:43:
[image]

Szybet, 2025-07-30:
[image]

This is the most interesting issue till now

well, more logs needed
- Go to config.h, enable DEBUG to 1, open serial, try to cause and record any interesting logs / errors
- if that doesn't help, go to https://github.com/Szybet/InkWatchy/blob/1776833704addac017ea91e617e3890f5982f77d/resources/tools/buildTime/sdkconfigs/sdkconfig_general.defaults#L117-L125 and disable the "None" options, enable warning ones
- if that doesnt show anything, modify these to show more logs https://github.com/Szybet/InkWatchy/blob/1776833704addac017ea91e617e3890f5982f77d/resources/tools/buildTime/sdkconfigs/sdkconfig_general.defaults#L14-L16

Szybet, 2025-07-30:
Possible fix: in config.h change #define FONT_COUNT 30 to 10

Szybet, 2025-07-30:
Regenerate resources, show output

Then uploud them, replicate, look for logs

TheSimpleZ, 2025-07-31:
After we spoke on discord, and you made a fix for the rust things crashing, it seems like I can't reproduce this anymore. Closing...