Szybet/InkWatchy #58: cannot connect to wifi, RST_PANIC or hang

picarica, 2025-08-07:
so this has been happening randomly for some months, but mostly it connects for like 4th try, but now it has not connected for 2 weeks.

[example of a hanging connect](https://litter.catbox.moe/688acmvneh7n2nc6.mp4)

here is example of it crashing, this crash happend after it was hung up for like 7 minutes, until it threw me back to home screen
https://github.com/user-attachments/assets/2ecb4d45-14ae-4698-9c40-ed489eb875d4

here are my configs, not sure if i have something wrong, it stopped working without update
[confidential.log](https://github.com/user-attachments/files/21659219/confidential.log)
[config.log](https://github.com/user-attachments/files/21659218/config.log)

Szybet, 2025-08-07:
Your config is outdated, the crash is caused by TASK_STACK_WIFI being outdated (Just use a diff tool to compare...)

yes, connecting can be slow but it's the only was to ensure it works for everyone. Disable power saving features like HARDWARE_POWER_SAVINGS and SOFT_START_THINGS and increase speed by CPU_SPEED (you can also tinker with the wifi code itself, idk set more power to the antena)

wifi connects only to 2.4Ghz networks, hardware limitation

also dont use WIFI_COUNTRY_FIX if it's not needed (if it works without it)

Szybet, 2025-08-07:
Also what watchy version do you have, v2?

picarica, 2025-08-07:
```
Checking "python3" ...
Python 3.12.3
"python3" has been detected

ERROR: ESP-IDF Python virtual environment "/root/.espressif/python_env/idf5.4_py3.12_env/bin/python" not found. Please run the install script to set it up before proceeding.
./getCoreDump.sh: line 23: idf_tools.py: command not found
Checking "python3" ...
Python 3.12.3
"python3" has been detected

ERROR: ESP-IDF Python virtual environment "/root/.espressif/python_env/idf5.4_py3.12_env/bin/python" not found. Please run the install script to set it up before proceeding.
Env name found in inkwatchy env file in /tmp/: Watchy_2
./getCoreDump.sh: line 28: /root/.espressif/python_env/idf5.4_py3.12_env/bin/esp-coredump: No such file or directory

```

ermmm

Szybet, 2025-08-07:
bruh wait i will fix it

Szybet, 2025-08-07:
I pushed a fix to your fork, pull, try again

picarica, 2025-08-07:
oh yeah, i am working from your repo, not mine, ill compile from mine

Szybet, 2025-08-07:
> again? i didnt get any text

what, whatever record it (terminal)

Szybet, 2025-08-07:
as in simply launch coredump again after pulling the changes i pushed

picarica, 2025-08-07:
```
INFO: Using IDF_PATH '/workspaces/InkWatchy/.root/.platformio/packages/framework-espidf' for installation.
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32c6, esp32s2, esp32, esp32h2, esp32c2, esp32s3, esp32c5, esp32c61, esp32p4, esp32c3
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping riscv32-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20250422 (already installed)
Skipping esp-rom-elfs@20241011 (already installed)
Installing Python environment and packages
Python 3.12.3
pip 25.2 from /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)
Skipping the download of /root/.espressif/espidf.constraints.v5.4.txt because it was downloaded recently.
Upgrading pip...
Requirement already satisfied: pip in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (25.2)
Upgrading setuptools...
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (71.0.0)
Installing Python packages
 Constraint file: /root/.espressif/espidf.constraints.v5.4.txt
 Requirement files:
  - /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 7)) (71.0.0)
Requirement already satisfied: packaging in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 8)) (25.0)
Requirement already satisfied: click in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 11)) (8.1.8)
Requirement already satisfied: pyserial in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 12)) (3.5)
Requirement already satisfied: cryptography in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 13)) (44.0.3)
Requirement already satisfied: pyparsing in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 14)) (3.2.3)
Requirement already satisfied: pyelftools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 15)) (0.32)
Requirement already satisfied: idf-component-manager in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 16)) (2.2.2)
Requirement already satisfied: esp-coredump in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 17)) (1.14.0)
Requirement already satisfied: esptool in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 18)) (4.9.0)
Requirement already satisfied: esp-idf-kconfig in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 19)) (2.5.0)
Requirement already satisfied: esp-idf-monitor in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 20)) (1.7.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 21)) (0.1.9)
Requirement already satisfied: esp-idf-size in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (li [... cut ...]
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision v1.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 90:15:06:e6:61:04
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
29508 (100 %)
29508 (100 %)
Read 29508 bytes at 0x003f0000 in 2.7 seconds (88.5 kbit/s)...
Hard resetting via RTS pin...

Failed to load core dump: Invalid application image for coredump: coredump SHA256(df4cc1d9a) != app SHA256().
```

Szybet, 2025-08-07:
You need to crash it with the same firmware you have on your pc at the same time, then retrieve the coredump

Szybet, 2025-08-07:
It's noted on the wiki: Don't recompile / change config.h then recompile - You need the same build on your PC as on your device

;p

picarica, 2025-08-07:
okay but its happening anyways, i did first 

1. fresh clone,
2. copy over config.h
3. first build, 
4. flash
5. sync with wifi on watch
6. get core dump
7.  still getting **Failed to load core dump: Invalid application image for coredump: coredump SHA256(94a6d8ed9) != app SHA256().**

am i stupid? fuck

i even did 

1. pio run -e $ENVNAME && pio run -t upload -e $ENVNAME
2. PLATFORMIO_ENV_NAME=$ENVNAME ./resources/tools/other/tasks/taskerDebug.sh

still error **Failed to load core dump: Invalid application image for coredump: coredump SHA256(94a6d8ed9) != app SHA256().**

Szybet, 2025-08-07:
Did in between 5 and 6 it crash?

picarica, 2025-08-07:
yes

Szybet, 2025-08-07:
go to resources/tools/other/debug/getCoreDump.sh change the first line to `#!/bin/bash -x` then run coredump again, copy the output

picarica, 2025-08-07:
```
+ source ../../globalFunctions.sh
+ cd ../../../../
+ .root/.platformio/packages/framework-espidf/install.sh
INFO: Using IDF_PATH '/workspaces/InkWatchy/.root/.platformio/packages/framework-espidf' for installation.
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /root/.espressif/idf-env.json
Selected targets are: esp32, esp32p4, esp32c3, esp32c5, esp32c2, esp32s3, esp32s2, esp32h2, esp32c61, esp32c6
Current system platform: linux-amd64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping riscv32-esp-elf@esp-14.2.0_20241119 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20250422 (already installed)
Skipping esp-rom-elfs@20241011 (already installed)
Installing Python environment and packages
Python 3.12.3
pip 25.2 from /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages/pip (python 3.12)
Skipping the download of /root/.espressif/espidf.constraints.v5.4.txt because it was downloaded recently.
Upgrading pip...
Requirement already satisfied: pip in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (25.2)
Upgrading setuptools...
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (71.0.0)
Installing Python packages
 Constraint file: /root/.espressif/espidf.constraints.v5.4.txt
 Requirement files:
  - /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 7)) (71.0.0)
Requirement already satisfied: packaging in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 8)) (25.0)
Requirement already satisfied: click in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 11)) (8.1.8)
Requirement already satisfied: pyserial in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 12)) (3.5)
Requirement already satisfied: cryptography in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 13)) (44.0.3)
Requirement already satisfied: pyparsing in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 14)) (3.2.3)
Requirement already satisfied: pyelftools in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 15)) (0.32)
Requirement already satisfied: idf-component-manager in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 16)) (2.2.2)
Requirement already satisfied: esp-coredump in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 17)) (1.14.0)
Requirement already satisfied: esptool in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 18)) (4.9.0)
Requirement already satisfied: esp-idf-kconfig in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 19)) (2.5.0)
Requirement already satisfied: esp-idf-monitor in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 20)) (1.7.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in /root/.espressif/python_env/idf5.4_py3.12_env/lib/python3.12/site-packages (from -r /workspaces/InkWatchy/.root/.platformio/packages/framework-espidf/tools/requirements/requirements.core.txt (line 21)) (0.1.9)
[... cut ...]
Chip is ESP32-PICO-D4 (revision v1.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 90:15:06:e6:61:04
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
29700 (100 %)
29700 (100 %)
Read 29700 bytes at 0x003f0000 in 2.7 seconds (88.7 kbit/s)...
Hard resetting via RTS pin...

Failed to load core dump: Invalid application image for coredump: coredump SHA256(94a6d8ed9) != app SHA256().

```
this should be it

Szybet, 2025-08-07:
> $ head -n 5 resources/tools/other/debug/getCoreDump.sh
> 
> ```
> #!/bin/bash -x
> 
> # Nevermind!
> # In .esp-idf/espressif/python_env/idf5.1_py3.12_env/lib64/python3.12/site-packages/esp_coredump/corefile/loader.py
> # Or if it doesn't exist .root/.espressif/python_env/idf5.3_py3.12_env/lib/python3.12/site-packages/esp_coredump/corefile/loader.py
> ```

No, don't do it, the wiki says to not do it. Read the wiki

picarica, 2025-08-07:
just showed i added the -x :D

picarica, 2025-08-07:
worked fine i guess, lets goo 
```

/dev/ttyS0 failed to connect: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Serial port /dev/ttyACM0
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision v1.1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 90:15:06:e6:61:04
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
29700 (100 %)
29700 (100 %)
Read 29700 bytes at 0x003f0000 in 2.7 seconds (89.2 kbit/s)...
Hard resetting via RTS pin...

===============================================================
==================== ESP32 CORE DUMP START ====================
The ROM ELF file won't load automatically since it was not found for the provided chip type.

Crashed task handle: 0x3ffdd7d4, name: 'loopTask', GDB name: 'process 1073600468'
Crashed task is not in the interrupt context
Panic reason: assert failed: void rustPanic() rEGenegic.cpp:23 (0)

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x4012dd66
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40083cfc          0x40083cfc <panic_abort+16>
lbeg           0x4000c28c          1073791628
lend           0x4000c296          1073791638
lcount         0x0                 0
sar            0x10                16
ps             0x60a20             395808
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80091a6c          -2146887060
a1             0x3ffe6a40          1073637952
a2             0x3ffe6a80          1073638016
a3             0x2                 2
[... 314 lines cut ...]
.coredump.tasks.data 0x3ffebce0 0x210 RW 
.coredump.tasks.data 0x3ffdbfcc 0x154 RW 
.coredump.tasks.data 0x3ffdbe00 0x1a0 RW 
.coredump.tasks.data 0x3ffdc130 0x154 RW 
.coredump.tasks.data 0x3ffde9e0 0x1b0 RW 
.coredump.tasks.data 0x3ffbd9c0 0x154 RW 
.coredump.tasks.data 0x3ffdb750 0x1b0 RW 
.coredump.tasks.data 0x3ffb795c 0x154 RW 
.coredump.tasks.data 0x3ffb7770 0x1c0 RW 
.coredump.tasks.data 0x3ffeceac 0x154 RW 
.coredump.tasks.data 0x3ffecc70 0x230 RW 
.coredump.tasks.data 0x3ffee014 0x154 RW 
.coredump.tasks.data 0x3ffedd50 0x2b0 RW 
.coredump.tasks.data 0x3ffb71f4 0x154 RW 
.coredump.tasks.data 0x3ffb7000 0x1c0 RW 
.coredump.tasks.data 0x3ffef7a0 0x154 RW 
.coredump.tasks.data 0x3ffef590 0x200 RW 
.coredump.tasks.data 0x3ffbd584 0x154 RW 
.coredump.tasks.data 0x3ffbd3b0 0x1c0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!

```

Szybet, 2025-08-07:
Show me your config.h

Try with a clean config.h

explain to me exactly how to replicate, exactly

picarica, 2025-08-07:
```
===============================================================
==================== ESP32 CORE DUMP START ====================
The ROM ELF file won't load automatically since it was not found for the provided chip type.

Crashed task handle: 0x3ffdd7d4, name: 'loopTask', GDB name: 'process 1073600468'
Crashed task is not in the interrupt context
Panic reason: assert failed: void rustPanic() rEGenegic.cpp:23 (0)

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x4012dd66
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40083cfc          0x40083cfc <panic_abort+16>
lbeg           0x4000c28c          1073791628
lend           0x4000c296          1073791638
lcount         0x0                 0
sar            0x10                16
ps             0x60220             393760
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80091a6c          -2146887060
a1             0x3ffe6a40          1073637952
a2             0x3ffe6a80          1073638016
a3             0x2                 2
a4             0xa                 10
a5             0x3ffe6fcc          1073639372
a6             0x18                24
a7             0x3ffe6fb4          1073639348
a8             0x3ffd3804          1073559556
a9             0x1                 1
a10            0x28                40
a11            0x3ffe6b77          1073638263
a12            0x0                 0
a13            0x3fffeed0          1073737424
a14            0x3ffe70c4          1073639620
a15            0x800               2048

==================== CURRENT THREAD STACK =====================
#0  panic_abort (details=0x3ffe6a80 "assert failed: void rustPanic() rEGenegic.cpp:23 (0)") at /root/.platformio/packages/framework-espidf/components/esp_system/panic.c:481
#1  0x40091a6c in esp_system_abort (details=0x3ffe6a80 "assert failed: void rustPanic() rEGenegic.cpp:23 (0)") at /root/.platformio/packages/framework-espidf/components/esp_system/port/esp_system_chip.c:87
#2  0x40098da0 in __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /root/.platformio/packages/framework-espidf/components/newlib/assert.c:80
#3  0x400d9fe5 in rustPanic () at src/other/rustExport/rEGenegic.cpp:23
#4  0x4016c7b5 in __rustc::rust_begin_unwind ()
#5  0x4017307d in core::panicking::panic_nounwind_fmt ()
#6  0x40171d45 in __rustc::__rdl_oom ()
#7  0x40171a1f in __rustc::__rust_alloc_error_handler ()
[... 299 lines cut ...]
.coredump.tasks.data 0x3ffeced4 0x154 RW 
.coredump.tasks.data 0x3ffecc90 0x230 RW 
.coredump.tasks.data 0x3ffee03c 0x154 RW 
.coredump.tasks.data 0x3ffedd80 0x2b0 RW 
.coredump.tasks.data 0x3ffb71f4 0x154 RW 
.coredump.tasks.data 0x3ffb7000 0x1c0 RW 
.coredump.tasks.data 0x3ffbd584 0x154 RW 
.coredump.tasks.data 0x3ffbd3b0 0x1c0 RW 
.coredump.tasks.data 0x3ffef7c8 0x154 RW 
.coredump.tasks.data 0x3ffef5c0 0x200 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!

```

this one was right after the crash.

as seen in the video 

. but ye after replacing the config, it  doesnt crash, it just keeps connected until i turn it off 

[config.h.log](https://github.com/user-attachments/files/21664224/config.h.log)
not sure what was causing this in my config.h, i guess il update this config from fresh?

picarica, 2025-08-07:
and replicate, i just go to setting click wifi enable it and just wait,

Szybet, 2025-08-07:
Get a clean config then modify it so it connects to wifi (probably WIFI_COUNTRY_CODE?)

picarica, 2025-08-07:
it seems fine, it crashed once, but at least its syncing, it seems very weird, could be hardware issue. my watch is half broken halfy repaired, went trhoguht alot of abuse, i wonder for much are you selling yachtchys for ? with vibration motor

closed 2025-08-07 by picarica

Szybet, 2025-08-07:
details on discord server

Anyway, I'm not letting this issue go, it's not hardware

try with a clean config, if it still crashes, wait, I will give you an update, possible fix

Szybet, 2025-08-07:
I pushed a fix, try with it, with clean config, it's highly experimental

also, go to debug->hardware page and send me a picture

Szybet, 2025-08-07:
also, go to wifi page and dont exit it until it finishes

picarica, 2025-08-07:
i am on it

Szybet, 2025-08-07:
forgot to mention before it remove sdkconfig files in the main root dir of the repo, do platformio clean only then build

picarica, 2025-08-07:
okay i did full clean, clean config straightly copied from template, didnt flash resources but whatever, connecting fine, i keept it until it turned off and put me to home screen

[image]

Szybet, 2025-08-07:
I dont understand anything

latest changes, clean config that works (dont modify anything that is not needed for wifi to work)

does it crash?
does it connect to the wifi?
does it sync time?

it's normal it goes back to the homescreen after it fails to disconnect / or it synced

picarica, 2025-08-07:
yes its fine with clean config, with the one i sent, 3 attemts all success

[config.h.new-working.log](https://github.com/user-attachments/files/21667588/config.h.new-working.log)
this one, crashing happend

Szybet, 2025-08-07:
No worries, we can continue tommorow, I'm gonna try to replicate it

have a good day!

Szybet, 2025-08-07:
Also what is the watchface you use?

picarica, 2025-08-07:
yep, testing now

picarica, 2025-08-07:
okay, so far so good, 2 attempts, ill try more at home will update tomorrow if i dont forged xd

Szybet, 2025-08-07:
You enabled BLE, which eats memory even when it's not enabled (static buffers)

I pushed a small fix to improve it a bit (couldn't more)

but DO NOT ENABLE BLE UNTIL YOU WANT TO USE IT you can't, you don't have the app

also bitcoin was taking a lot of memory, but without BLE it's easily managable

i pushed a fix & merged the pull request, check it if it works now (DISABLE BLE)

picarica, 2025-08-08:
ye after your push, it has not crashed since i flashed it, tried at 2 different networks, all fine no crashes, even with BLE enable, very good. thank you

Szybet, 2025-08-08:
read the readme baiky app section

picarica, 2025-08-08:
ye i get it, but i cant seem to find the accompaying app, https://www.baiky.app/ is this it ?

picarica, 2025-08-11:
https://litter.catbox.moe/ing57fxqge7o4ix1.mp4
intresting it happend again, i thoguth i run out of memory but i just charged my watch (it was around 50%) and now it works, why would it fail (crash) with low charge?

so my testing, on friday completly fine
saturday fine, sunday fine
monday this crash.
i didnt flash new firmware, i didnt rebooted, weird behavior. chargin it made it better

picarica, 2025-08-11:
i am unable to replicate it after i charged it :P, ill try to get one when my watch gets discharged or will keep trying it

Szybet, 2025-08-11:
WAIT WAIT WAIT WAIT

first, wtf is going on with your screen

second, look

[image]

It's not PANIC, it's BROWNOUT which means voltage dropped too low, which matches that the watch was discharged a lot, wifi might have killed it further, simply charge it and dont launch wifi if its discharged a lot