Szybet/InkWatchy #60: failed to wait on curl `Multi` | curl calendar

Meffi29a, 2025-08-10:
No Luck with my calendar, need help.. again

i am able to run curl to the url in the docker terminal. I see the whole file .
It is an iCloud-URL. I do not know if every caldav is that messy. I have no other calendar to try.

the curl version is some Outdated? 
the repo says 8.15.0 ist out. Rust is using 8.5.0

No idea how to upgrade (even no idea if needed)
Never used rust, never used Cargo. Is to much new stuff in this nice project.

```
Found URLs:
https://p122-caldav.icloud.com/published/2/xxxNoKeyForPublic
error: failed to wait on curl `Multi`

Caused by:
  Unrecoverable error in select/poll
cat: 'resources/tools/fs/littlefs/calendar/index.txt*': No such file or directory
./getFs.sh: line 7: ../other/in/esptool: No such file or directory
./getFs.sh: line 9: ./in/mklittlefs: No such file or directory
./flashPersonalFs.sh: line 6: ./in/mklittlefs: No such file or directory
./flashPersonalFs.sh: line 7: ../other/in/esptool: No such file or directory

 *  The terminal process "/bin/bash '-c', 'resources/tools/other/tasks/taskerOther.sh'" terminated with exit code: 127. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

```
oot@71f80285ee6c:/workspaces/InkWatchy/resources/tools/fs/littlefs/calendar# curl --version
curl 8.5.0 (x86_64-pc-linux-gnu) libcurl/8.5.0 OpenSSL/3.0.13 zlib/1.3 brotli/1.1.0 zstd/1.5.5 libidn2/2.3.7 libpsl/0.21.2 (+libidn2/2.3.7) libssh/0.10.6/openssl/zlib nghttp2/1.59.0 librtmp/2.3 OpenLDAP/2.6.7
Release-Date: 2023-12-06, security patched: 8.5.0-2ubuntu10.6
Protocols: dict file ftp ftps gopher gophers http https imap imaps ldap ldaps mqtt pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS brotli GSS-API HSTS HTTP2 HTTPS-proxy IDN IPv6 Kerberos Largefile libz NTLM PSL SPNEGO SSL threadsafe TLS-SRP UnixSockets zstd
root@71f80285ee6c:/workspaces/InkWatchy/resources/tools/fs/littlefs/calendar# 
```

picarica, 2025-08-11:
ye if you can provide the ics link privately i can look into it too,  i never worked with icloud so i dont have much clue with curl is failing
https://github.com/rust-lang/cargo/issues/11435#issuecomment-1740163332
but this seems like a viable fix, note more of a curl issue but more cargo issue, can you try setting 
`ulimit -n 1024` ?
i have no idea how that can help :D but you can try it out

Szybet, 2025-08-11:
Hmmm yes worth trying but I doubt

Szybet, 2025-08-11:
works in my devcontainer

but other things work for you? Like downloading platformio inside of devcontainer? Idk weird

Anyway, macOS issue, can't help

Meffi29a, 2025-08-11:
> works in my devcontainer
> 
> but other things work for you? Like downloading platformio inside of devcontainer? Idk weird
> 
> Anyway, macOS issue, can't help

Looks like an macOS issue ore a cargo script thing. Have no knowledge to point the one ore the other.

The only problems up to now are -the serial connection

The setup worked without any problems. Just as the tutorial.

Up to now I didn't tested the OTA programming. Only updated with a script compiled DEMO. Manually flashed in the macOS-environment. 
Should test the OTA, as it would be nice to know for Others.

Szybet, 2025-08-11:
OTA what? There is no OTA for InkWatchy

Meffi29a, 2025-08-11:
> Also some comments from confidential.h
> ```
> // If you want just a path, implement it yourself, look up the script and the program arguments
> // Link to your ICS file to download it
> ```
> So contributing is welcome

You pointing the right lines. I try to give a least some adds to the user guide. So I may help others to find their way. At the moment I try to understand the project itself. Very clean, but bit frustrating. Very much new technology for now.

Meffi29a, 2025-08-11:
> OTA what? There is no OTA for InkWatchy

Sorry, there is no OTA. I thought there was one provided by platformIO...

Szybet, 2025-08-11:
Nope, InkWatchy is too big to support that (or in other words, watchy manufacturers are too cheap to add 20 cents to the price for 8MB)

Meffi29a, 2025-08-11:
OK, I've been poking around for a while and I've finally gotten there. The error "failed to wait on curl Multi" was because Cargo wasn't compiled. 
After 
manually running the command 
"cargo run --manifest-path src/ui/places/calendar/ics-to-json-rs/Cargo.toml -- -u "$url" -o $calendar_dir/",
lot of things were built.

The error was probably telling me that the build process was taking too long because a lot of dependencies had to be created. Now it works with my URL.

closed 2025-08-11 by Meffi29a