Szybet/InkWatchy #63: Watchdog Issue on V3

CoryRamsey, 2025-08-16:
Hello. I've just received a new V3 watchy. I'm still working on wrapping my head around the codebase here, but I'm having an issue right out of the box with your pre-built binary and, as expected, the version I'm compiling myself. There seems to be an issue with the watchdog. It never makes it to the watch face after flashing and resetting. An additional reset just sends it back into a panic.

I'm poking at it a bit to try to debug. I've managed to get it to actually boot with some modifications, but I'm still getting random panics during use.

Here is my core dump on completely unmodified code:

===============================================================
==================== ESP32 CORE DUMP START ====================
The ROM ELF file won't load automatically since it was not found for the provided chip type.

Crashed task handle: 0x3fcb9ec0, name: 'watchdogTask', GDB name: 'process 1070309056'
Crashed task is not in the interrupt context
Panic reason: assert failed: void loopWatchdogTask(void*) watchdogTask.cpp:53 (0)

================== CURRENT THREAD REGISTERS ===================
exccause       0x0 (IllegalInstructionCause)
excvaddr       0x0
epc1           0x42044d9a
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x403783d8          0x403783d8 <panic_abort+16>
lbeg           0x40056f08          1074097928
lend           0x40056f12          1074097938
lcount         0x0                 0
sar            0x8                 8
ps             0x60420             394272
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x8037ee14          -2143818220
a1             0x3fcc11e0          1070338528
a2             0x3fcc1220          1070338592
a3             0x1                 1
a4             0xa                 10
a5             0xc11b0000          -1055195136
a6             0x60023             393251
a7             0x3                 3
a8             0x3fcaf994          1070266772
a9             0x1                 1
a10            0x3fcc1317          1070338839
a11            0x3fcc1317          1070338839
a12            0x3fcc1317          1070338839
a13            0x3fca83c0          1070236608
a14            0x3fca6178          1070227832
a15            0x3fcaffdc          1070268380

==================== CURRENT THREAD STACK =====================
#0  panic_abort (details=0x3fcc1220 "assert failed: void loopWatchdogTask(void*) watchdogTask.cpp:53 (0)") at /root/.platformio/packages/framework-espidf/components/esp_system/panic.c:481
#1  0x4037ee14 in esp_system_abort (details=0x3fcc1220 "assert failed: void loopWatchdogTask(void*) watchdogTask.cpp:53 (0)") at /root/.platformio/packages/framework-espidf/components/esp_system/port/esp_system_chip.c:87
#2  0x40386c3c in __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /root/.platformio/packages/framework-espidf/components/newlib/assert.c:80
[... 142 lines cut ...]
.coredump.tasks.data 0x3fcb9ec0 0x154 RW 
.coredump.tasks.data 0x3fcc1120 0x3a0 RW 
.coredump.tasks.data 0x3fcba424 0x154 RW 
.coredump.tasks.data 0x3fcba190 0x280 RW 
.coredump.tasks.data 0x3fcba88c 0x154 RW 
.coredump.tasks.data 0x3fcba600 0x280 RW 
.coredump.tasks.data 0x3fcbf3f8 0x154 RW 
.coredump.tasks.data 0x3fcbed70 0x680 RW 
.coredump.tasks.data 0x3fcb8ba0 0x154 RW 
.coredump.tasks.data 0x3fcb88f0 0x2a0 RW 
.coredump.tasks.data 0x3fcbaff4 0x154 RW 
.coredump.tasks.data 0x3fcbad90 0x250 RW 
.coredump.tasks.data 0x3fcb541c 0x154 RW 
.coredump.tasks.data 0x3fcb5150 0x2a0 RW 
.coredump.tasks.data 0x3fcb9d5c 0x154 RW 
.coredump.tasks.data 0x3fcc0e30 0x290 RW 
.coredump.tasks.data 0x3fcb9a64 0x154 RW 
.coredump.tasks.data 0x3fcc09a0 0x290 RW 
.coredump.tasks.data 0x3fcb4cb4 0x154 RW 
.coredump.tasks.data 0x3fcb49e0 0x2a0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================

Let me know if I'm completely missing something dumb. I really appreciate your work on this project. I hope to get this resolved. Thanks!

Szybet, 2025-08-16:
There are issues with your accelerometer?

Anyway, I pushed a small improvement, pull it, try it. Also did the accelerometer work with other firmwares? As in were steps counted?

Some more things to try:
1. FORCE_DISABLE_ACC to 1
2. WATCHDOG_TASK to 0
(Don't combine them)

izzylaif, 2025-09-03:
naturlich, that's why I commented.
You may need to guide me through the log collecting part since I'm very new to arduino stuff. can the logs be downloaded through web or platfromio?

Szybet, 2025-09-03:
Cool :D

Hmm first I propose you follow the inkwatchy setup in the wiki if you haven't already. Then You can follow the instructions written in this issue of what to test, then provide logs. Feel free to ask if you don't know something

Logs are gathered via platformio in a regular platformio way (so serial monitor)

I will prepare a modified binary program for you to try to flash to the v3, I suspect this will give me answers, but the steps above are still worth trying

izzylaif, 2025-09-03:
I can confirm this one works!

PS: I drop you a PM, please read

Szybet, 2025-09-03:
Thanks!

Well that's sad that it works, I need to roll back a frameworks update. Expect a full fix later today

PM where? I don't see anything