Szybet/InkWatchy #67: [Suggestion] Optional Dark Mode

DomainOfThePublic, 2025-08-19:
I was encouraged to make a suggestion here. It would be nice to have an option that would invert the colours of either the entire system, or just the watchface, depending on user preference (with a config.h setting for chosen default state, as well as a toggle in settings while the watch is live)

Szybet, 2025-08-19:
Done, via INITIAL_SCREEN_INVERT and WATCHDOG_INVERT_COLORS in config.h (to update config.h, look: https://github.com/Szybet/InkWatchy/wiki/Updating-the-firmware

Only config.h options because I don't see a reason for it to be changeable at runtime, like, how often do you change dark / white mode on your phone. Anyway, feel free to add this option yourself :D

DomainOfThePublic, 2025-08-25:
I should have tested this earlier, I'm sorry for the delay. Charts do not cooperate with dark mode due to hardcoded colours from the Grafici-GFX library (screen shows as full black with inverted colours, due to SCWhite being used to fill the background, while "black" is pulled from the library)

I have no idea if there's an easy fix for this to allow for dark mode without having to mess with the library itself, so it might instead be preferable to **return the two instances of SCWhite in chart.cpp to GxEPD_WHITE** to force light mode in the weather charts regardless of user setting

closed 2025-08-26 by Szybet